taborlab/LPA-hardware #14: File format

RaphP, 2017-03-08:
Dear Taborlab,

we are very interested in your project, thank you very much for uploading it on GitHub !!
I personally work mostly on open source software when I can. Would it be possible to have generic 3D printed files (.STL for exemple) instead of solid works ? 
We might have a solid work licence at the lab, so maybe I can have the tech guy that use it let me convert the files and then with your permission I can upload them on the git ?

Best regards,
Raphaël.

castillohair, 2017-03-08:
Hi Raphael,

Please find the .STL files [here](https://github.com/taborlab/LPA-hardware/releases/download/v1.0/lpa.zip). We keep these, as well as other ready-for-manufacturing files in the [Release](https://github.com/taborlab/LPA-hardware/releases) section.

Hope this helps,

Sebastian.

closed 2017-03-08 by castillohair

RaphP, 2017-03-08:
Dear Sebastian,

thank you very much for that !

Best regards,
Raphaël.