taborlab/LPA-hardware #15: Outsourced pcb manufacturing + assembly

cuong450, 2017-06-16:
Here are the issues mentioned by the outsourced company:

Thank you so much for your updated design, and we still have below issues:

1.#11_L1 p/n: VLS6045EX-151M does the orientation matter for this part?
 
[issue1_2 jpg](https://user-images.githubusercontent.com/15172848/27229032-fa494398-5277-11e7-9038-b892934466d1.jpeg)
[image]

2. In your new Gerber, there is no negative pole marks any more. Please comment.(We know you don't need us the solder the parts on the top side, but we just remind.)

[image]

KarlGerhardt, 2017-06-16:
Hi Cuong,

1. I believe inductors in general don't have polarity, but each pad of the component should be aligned and soldered with only a single pad on the PCB. This would be consistent with having the component's indicator "dot" over one pad or the other. Sebastian can you confirm?  

2. The orientation does matter. It should be visible in the front silkscreen as in the below image.

[image]

Cheers,
Karl

castillohair, 2017-06-16:
Hi Coung,

1. As Karl said, the orientation of the inductor L1 does not matter.

2. I'm opening the gerber files directly and the polarity marks for the LEDs are there, specifically in file `lpa-F.SilkS.gto`. I used the program Gerbv version 2.6A, which is free to download. Screenshot below.

[image]

I'm worried that if the manufacturing company can't see certain layers for some reason, then they might miss the polarity in certain components in which it matters. Can you check with the company that they can see the polarity in capacitors C3, C5, diode D49, and component X1?

Hope this helps.

castillohair, 2017-06-20:
Cuong,

Did you ask about the polarity of the other components?

castillohair, 2017-06-25:
Closed due to inactivity.

closed 2017-06-25 by castillohair