taborlab/LPA-hardware #17: Power Supply

apolnaszek, 2017-06-23:
Hello, I recently started working on a copy of the LPA and was wondering what voltage power supply the PCB needs. I didn't see a power supply on the BOM.

apolnaszek, 2017-06-23:
Looking through the data sheets I think I've come to the conclusion that a 12V, 2.1mm x 5.5mm, AC/DC power supply is needed. Can anyone confirm?

KarlGerhardt, 2017-06-23:
Hello apolnaszek,
Glad to hear you are building an LPA. Definitely read through the supplemental information to the publication. Table S1 shows the parts (beyond the BOM for circuit board assembly) you will need, including the power supply. You will need a switching power supply which supplies 5V DC. 

Best regards,
Karl

closed 2017-06-23 by KarlGerhardt

apolnaszek, 2017-06-23:
Thanks for the quick response, Karl. I'm new to GitHub and I've looked through the files and folders a couple times but couldn't find Table S1. Could you point it out for me?

KarlGerhardt, 2017-06-23:
Sorry, I assumed you were familiar with the publication on this hardware. 
https://www.nature.com/articles/srep35363 
Near the bottom you can find the supplementary information as a .pdf

apolnaszek, 2017-06-27:
I've been looking through the publication the last few days and trying to come up with a list of parts to order. I'm a little confused as to how the S1 table and BOM overlap. Some parts are on both lists and some aren't. Also, both lists have a different 16 MHz resonator crystal. Do I need both or just one?

KarlGerhardt, 2017-06-28:
The BOM only has the components for commercial assembly of the circuit board. Table S1 should have ALL necessary electrical components. However, the current release (see "Releases") has an updated circuit board (with some new/different components). These changes have been made in the BOM, but we have not released an updated Table S1 (or any updated documentation documentation for that matter).

If you want to build the new board, you will want to use the components in the BOM. Keep in mind, the BOM doesn't include the board components which we recommend soldering yourself (e.g. LED sockets), but there have been no changes to those.

castillohair, 2017-06-28:
To summarize, you will need all components specified in the BOM, plus LED sockets (48 units, PCV_220-ND on digikey) and a power supply (418-TRG1505-V on mouser).