taborlab/LPA-hardware #18: reference designators (Y1,C4) in the pick and place file

cuong450, 2017-06-26:
There are no coordinate for below reference designators (Y1,C4) in the pick and place file. Could you please help resend us the complete pick and place file?

[image]

KarlGerhardt, 2017-06-26:
These are components from the original design and are no longer used. The complete pick and place file is called "LPA_component.xlsx" located in the production files folder. They seem to be generating their own list based on the gerbers, which probably contains some vestigial links to old components. Sebastian can you comment?

castillohair, 2017-06-26:
I manually removed every trace of old components from the KiCAD files (they're essentially text files), which were then used to generate the gerber files. You can open file `lpa.kicad_pcb` with a text editor and search for `Y1`, and you are not gonna find it. Same with each one of the gerber files and the drill file. So I don't think this is the case. My best guess is that they just modified a component file that they started from the previous design, instead of starting from scratch as they should have.

I suggest you email them the file "LPA_component.xlsx" from the production folder in the **latest** design. If they are somehow generating the component file from the gerber files, tell them to do it from scratch.

closed 2017-06-28 by KarlGerhardt