taborlab/LPA-hardware #19: Atmel Connectivity Issue at 2.3V rather than 3.3V

cuong450, 2017-07-28:
Hi,
I am currently having issues connecting to Atmel studio, my target voltage is 2.3V rather than the instructed 3.3V. 

My board was manufactured and soldered by an outsourced company. I have not manually soldered the LED sockets on, but I have manually soldered on the barrel connector, the reset switch (is this orientation specific?), the 3 ON ERR FIN indicator LEDs  ( orientation of cathodes pointing toward the ON ERR FIN etching ). Please let me know what I can do to trouble shoot the issue? Thank you.

[image]
[image]
[image]

castillohair, 2017-07-28:
Hello,

I have a couple of follow up questions.

- Is 2.3 the voltage according to Atmel Studio?
- What power supply are you using?
- I would need a closer look at the programming pin connectors. It is currently hard to see if the wires match the respective pins, especially the red wire.
- Is this a problem with all the boards that you received? or just that one?
- The logical next step would be to check voltages at different points in the board and isolate the problem. Do you have access to a multimeter? Do you feel confident in using it? I ask because if you're not experienced and you're not careful, you could short two board contacts with the probes and burn something. 
- The last option is for you to send us the board and we can look at it here and mail it back (all mailing costs covered by you or your lab, of course).e

cuong450, 2017-08-01:
- Is 2.3 the voltage according to Atmel Studio?
yes reading is through Atmel Studio 7.0
- What power supply are you using?
5.0V
- I would need a closer look at the programming pin connectors. It is currently hard to see if the wires match the respective pins, especially the red wire.
The red wire is facing facing towards the reset button and away from SD card
- Is this a problem with all the boards that you received? or just that one?
I did not go through with the purchase of other boards, I wanted to verify that this one works before I order more
- The logical next step would be to check voltages at different points in the board and isolate the problem. Do you have access to a multimeter? Do you feel confident in using it? I ask because if you're not experienced and you're not careful, you could short two board contacts with the probes and burn something.
Do you have a diagram of the expected voltages expected at each pin of the electrical board? If so, I can do the checks. If no please set up a Skype session so we can go through this together
- The last option is for you to send us the board and we can look at it here and mail it back (all mailing costs covered by you or your lab, of course).e
I would prefer to have an understanding of where the problem lays for future references, so I want to troubleshoot this in-house before the last resort.

castillohair, 2017-08-02:
The red wire should be coming into pin #1 of the connector, which is highlighted on the board with a white dot next to it. I can see the dot in your picture, below the P2 label. In other words, pin #1 of the connector is next to the SD card connector and away from the reset button, into the board and away from the edge. Please check these connections and let me know.

What programmer are you using?

cuong450, 2017-08-02:
Yes, this may be the issue. My current programmer (Atmel-ICE) . The red line does not go to the white dot on this model's configuration, the red line goes to a middle pin, which is incorrect.

https://www.digikey.com/product-detail/en/microchip-technology/ATATMEL-ICE-BASIC/ATATMEL-ICE-BASIC-ND/4753381

A collaborator provided me with this controller because they could not purchase the obsolete AVRISP mkII programmer (https://www.digikey.com/products/en?keywords=AVRISP%20mkII%20programmer). Do you have a source provider for this obsolete item?

castillohair, 2017-08-02:
I see. Your programmer should certainly work, but it also has many more advanced programming modes that we're not using and could be causing you some confusion.

Going through the manual, it seems that the red wire should go to pin 3, which the middle pin facing the inside of the board and away from the edge, which is kind of confusing. Please make sure that this is the case. Also, make sure that you select programming mode ISP or SPI when you try to connect to the programmer via Atmel Studio.

Let me know if any of this works.

cuong450, 2017-08-02:
Yes it works. I was under the impression that the red wire should be facing away from the SD card as instructed in your manual file, but this is not applicable (but rather opposite) for the ATMEL ICE programmer. Thanks will let you know about further progress.

castillohair, 2017-08-02:
Yeah apparently the red wire position depends on the programmer. It's good that we figured this out. Please keep us in touch about your progress.

I'll close this issue now.

closed 2017-08-02 by castillohair