taborlab/LPA-hardware #20: step G of Page 51 of 78 instruction

cuong450, 2017-08-03:
g. Make sure that the “Flash”, “Erase memory before programming”, and “Verify programmed content” checkboxes are activated, and click on the “Program” button. The messages “Erasing device... OK”, “Programming Flash...OK”, and “Verifying Flash...OK” should appear below.

[image]
[image]

Hi, Please see the attached screen shot of errors. What do you suggest?

castillohair, 2017-08-03:
How many times did you try? Sometimes programming just fails for no reason, but then it works the next time.

castillohair, 2017-08-03:
Also, on the fuses tab, you need to select the option with "16K CK/14CK" instead of "1K CK/14CK". This change is necessary for the new board model that you're building (as opposed to the model that we published in the paper).

cuong450, 2017-08-03:
after switching to "EXTXOSC_8MHZ_XX_16KCK_14CK_65MS"., I have tried it about 20 times now. In my past, certain SD cards are rejected by certain programming chips. Can you tell me the specific SD card brand that you are using? (I am using Kingston 16GB SD)

Do you have other suggestions?

castillohair, 2017-08-03:
So switching to "EXTXOSC_8MHZ_XX_16KCK_14CK_65MS" did not change anything?

cuong450, 2017-08-03:
It just worked with  "EXTXOSC_8MHZ_XX_16KCK_14CK_65MS" setting and removed SD card. Good to know

castillohair, 2017-08-03:
Glad it worked.

closed 2017-08-03 by castillohair

cuong450, 2017-08-03:
however, the error LED is on? what does this mean?

cuong450, 2017-08-03:
I see that it says it means no SD card available, but I plugged SD card back in. and error still present.

castillohair, 2017-08-03:
Did you push the reset button after inserting the SD card?

cuong450, 2017-08-03:
Yes, but the SD card does not have anything on it at the moment.

Am I suppose to put the dc.text and gcal.text file + the downloaded program.lpf files on the SD card?
Only those specific 3 files and nothing more?

In any folder of any name?

castillohair, 2017-08-03:
Yes. Those files should be in the SD card itself, not in a folder.