taborlab/LPA-hardware #23: problem with 96 well LPA

HMoerten, 2018-02-22:
Hi,

my name is Hanns-Martin &  and after reading your inspiring publication, I have successfully built the 24 well LPA. A little unusual though, since I am a molecular biologist by education, but  neither an optogeneticist nor very familiar with coding or with the details of electrotechnical circuitry. I have some experience with the development of biotechnical devices  though and was just enthusiastic about the possibility of building a small biotech device in the spirit of the DIY/ maker movement. 

After building the device, I simply asked a few labs here in Germany if they were interested of testing the LPA. To make a long story short, I was asked if I could come up with a 96 well solution as well. 

Since I am, as mentioned, not all too good at coding (though brave enough to change code here and there), I thought that daisychaining 12 TLC5941s would somehow be the simplest solution. After being able to programm .lpfs with the jupyter notebook  files, I found the 96well config.h you put on github and saw no reason why that should not work. On the hardware side, I constructed a breadboard circuit that mirrors the 24 well LPA, with more TLC5941 elements though.

So here comes my problem: I can get an 88 well version with 2 channels each (so 176 channels and TLC 5941_N set to 11) to work with. no problems so far, but whenever I switch to N set to 12 (and 192 channels) I get status LED errors, mainly of the "wrong SD file" type. Neither the compilation of the firmware.elf nor of the program.lpf files show errors. I tried all sorts of things, but have the impression there might be a physical barrier (buffer overflow? SPI limitations?). I have gone through the "demystifying the tlc 5940" book and found that there might be some clocking issues when daisychaining too many controllers.

So...any ideas on the subject? Any kind of help would be greatly appreciated!

Thank you from Cologne, Germany,

Hanns-Martin

castillohair, 2018-02-23:
Hi Hanns-Martin,

Good to know you're interested in our LPA! We have actually built a 96-well version in the past. See photos below:

[image]

[image]

However, we run into significant heating issues. It turns out that having that many LEDs and TLCs close together generates a lot of heat, which is unevenly distributed throughout the board, which in turn results in unpredictable cell growth rates and very noisy gene expression data. We tried several heat dissipation methods, including heatsinks (which you can see in the figure) and fans, but nothing worked to our satisfaction. Therefore, we never published on this.

I don't know what your firmware issues could be. I'm attaching below the code of the `config.h` file that we used for the 96-well device. It's been a while since we used it, so I can't guarantee that it will work with the latest firmware version. Remember that a `program.lpf` file for a 24-well LPA will not be recognized in a 96-well LPA, and this could give rise to a "wrong SD file" error. Also, as an electronic circuit gets faster and more complicated, board design becomes an important variable in terms of trying to aviod electromagnetic coupling. [This](http://www.analog.com/media/en/training-seminars/design-handbooks/Basic-Linear-Design/Chapter12.pdf) seems like a decent reference, although it is a little wordy. In short, a circuit with 12 TLCs might be beyond what can be achieved in a breadboard.

In summary, because of the overheating issues, I'd recommend to just stick with the 24-well version and avoid yourself some headaches.

```C
/*
 * config.h
 *
 * Created: 03/03/2014 02:29:05 PM
 *  Author: Sebastian Castillo
 */ 

// Global configuration variables
#define F_CPU 16000000UL

// Configuration for Tlc5941 module
#define Tlc5941_GSCLK_DDR DDRB
#define Tlc5941_GSCLK_PORT PORTB
#define Tlc5941_GSCLK_PIN PB0

#define Tlc5941_XLAT_DDR DDRB
#define Tlc5941_XLAT_PORT PORTB
#define Tlc5941_XLAT_PIN PB1

#define Tlc5941_SIN_DDR DDRD
#define Tlc5941_SIN_PORT PORTD
#define Tlc5941_SIN_PIN PD1

#define Tlc5941_MODE_DDR DDRD
#define Tlc5941_MODE_PORT PORTD
#define Tlc5941_MODE_PIN PD2

#define Tlc5941_BLANK_DDR DDRD
#define Tlc5941_BLANK_PORT PORTD
#define Tlc5941_BLANK_PIN PD3

#define Tlc5941_SCLK_DDR DDRD
#define Tlc5941_SCLK_PORT PORTD
#define Tlc5941_SCLK_PIN PD4

#define Tlc5941_TIMER 2

#define Tlc5941_USART_SPI 1

#define Tlc5941_N 12

#define Tlc5941_MANUAL_DC_FUNCS 1

#include <avr/io.h>
#include <avr/pgmspace.h>
#include "Tlc5941/Tlc5941.h"

/*static const Tlc5941_gsData_t well2channel[Tlc5941_numChannels] = {
	0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11,
	12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23,
	24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35,
	36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47,
	48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59,
	60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71,
	72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83,
	84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95,
	96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107,
	108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119,
	120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131,
	132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143,
	144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155,
	156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167,
	168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179,
	180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191};*/

static const Tlc5941_gsData_t well2channel[Tlc5941_numChannels] PROGMEM = {
	95, 94, 81, 80, 79, 78, 65, 64, 63, 62, 49, 48, 47, 46, 33, 32, 31, 30, 17, 16, 15, 14, 1, 0,
	93, 92, 83, 82, 77, 76, 67, 66, 61, 60, 51, 50, 45, 44, 35, 34, 29, 28, 19, 18, 13, 12, 3, 2,
	91, 90, 85, 84, 75, 74, 69, 68, 59, 58, 53, 52, 43, 42, 37, 36, 27, 26, 21, 20, 11, 10, 5, 4,
	89, 88, 87, 86, 73, 72, 71, 70, 57, 56, 55, 54, 41, 40, 39, 38, 25, 24, 23, 22, 9, 8, 7, 6,
	103, 102, 105, 104, 119, 118, 121, 120, 135, 134, 137, 136, 151, 150, 153, 152, 167, 166, 169, 168, 183, 182, 185, 184,
	101, 100, 107, 106, 117, 116, 123, 122, 133, 132, 139, 138, 149, 148, 155, 154, 165, 164, 171, 170, 181, 180, 187, 186,
	99, 98, 109, 108, 115, 114, 125, 124, 131, 130, 141, 140, 147, 146, 157, 156, 163, 162, 173, 172, 179, 178, 189, 188,
	97, 96, 111, 110, 113, 112, 127, 126, 129, 128, 143, 142, 145, 144, 159, 158, 161, 160, 175, 174, 177, 176, 191, 190,};
	
// Time step scaling
#define stepSizeScaling 1.0

```

HMoerten, 2018-02-23:
Hi Sebastian,

thank you very much for your immediate reply & the interesting information within!

I am aware of the heat problem & maybe using 12 TLCs is just too much. On the other hand, as mentioned, my technical problem is somewhere on the programming level between 11 and 12 TLCs and it s just driving me nuts. The config.h cannot be the problem since I have exactly/ principally  set them up (trying 48, 72, 88, well versions which all work fine) as the one you ve sent. As a control, I have used the one you ve sent though & get the same results  (192 channels just dont work) and simply reducing the number of channels bei 16 (and making N=11) combined with a program.lpf which also addresses 176 channels instead of 192 works absolutely fine. Crazy.   

Electromagnetic coupling would be a problem I m sure, especially when all the little cables on a breadboard setup serve as antennas, I think that interference phenomena using PWM is very likely. Might be different on a well designed PCB though. 

I will try a one channel version with 96 wells  (using only 6 TLCs) and in parallel maybe try using smd LEDs (losing the "switchability" of colors of course) which have their own little controllers (adafruit neopixels or the like). RGB Leds might be ok for some purposes, but definitely not for some wavelengths of course.

Thank you very much again for your help &

Kind regards

Hanns-Martin

closed 2018-04-04 by castillohair

tlnagy, 2019-03-21:
> However, we run into significant heating issues. It turns out that having that many LEDs and TLCs close together generates a lot of heat, which is unevenly distributed throughout the board, which in [...]

Over what time frame did you have the heating issues? I'm interested in building a 96 well device, but I'll probably only have 4-10 LEDs on at one time for about 10 seconds so I don't expect to have any issues with heating.

Is there a way that you could share the plans for the 96-well device?

castillohair, 2019-04-05:
We observed those issues over 8h experiments with the LEDs permanently ON at different intensities. I couldn't tell if your situation will not lead to heating until after trying. Nevertheless, our 24-well LPA design v1.1 has some upgrades that are supposed to minimize heating and that I would like to incorporate and test into the 96wp design.

Do you have an email so I can send you the 96-well design files?

tlnagy, 2019-04-05:
> Nevertheless, our 24-well LPA design v1.1 has some upgrades that are supposed to minimize heating and that I would like to incorporate and test into the 96wp design.

Great!

> Do you have an email so I can send you the 96-well design files?

My email is in my Github profile. Thanks!

castillohair, 2019-04-08:
No problem! I'm not in lab today, so I will send you the files tomorrow morning.