taborlab/LPA-hardware #27: 12pF Capacitor

phuongho43, 2019-02-18:
Seems like the listed 12pF capacitor is out of stock:
https://www.digikey.com/product-detail/en/samsung-electro-mechanics/CL21C120FBANNNC/1276-2579-1-ND/3890665

Would this one work?
https://www.digikey.com/product-detail/en/samsung-electro-mechanics/CL10C120FB8NNNC/1276-2175-1-ND/3890261

KarlGerhardt, 2019-02-18:
I don't think the replacement you suggested will work since the package size is different (2.00mm x 1.25mm vs. 1.60mm x 0.80mm).

This one ([720-1704-1-ND](https://www.digikey.com/product-detail/en/vishay-vitramon/VJ0805A120GXAPW1BC/720-1704-1-ND/8324124)) will probably work. The only difference I see is the tolerance is a little bit higher at 2%. 

Look good to you Sebastian?

castillohair, 2019-03-04:
Sorry for the delay. I've been out of commission as @KarlGerhardt knows.

Are these the capacitors for the crystal circuit? If that's the case, @KarlGerhardt 's choice is probably good. However, I would prefer to maintain the tolerance for such a sensitive portion of the design. This could be a better replacement: https://www.mouser.com/ProductDetail/AVX/08055U120FAT2A?qs=sGAEpiMZZMs0AnBnWHyRQKdiqyDPVQdAQn0ajLazkVM%3D. But I must say that I know very little about crystal oscillators, and it would take me a non-trivial amount of time to figure out whether the 1% increased tolerance matters or not. But worst case scenario, I think the timing error will shift by no more than a few seconds for an 8-hour program.

tldr; @phuongho43 if you're already bought @KarlGerhardt 's suggestion, don't worry and carry on. But if not, I would recommend to buy the one I suggested instead.

castillohair, 2019-03-07:
I opened a new issue (#28) to remind us to change the capacitors in the parts list.

I will now close this issue. @phuongho43 feel free to reopen it if you have further questions.

closed 2019-03-07 by castillohair