taborlab/LPA-hardware #32: SD card not recognized by LPA

GolgiWhillikers, 2019-03-08:
I am building an LPA v1.1. I have been unable to get the board to recognize the SD card (stable ERR light). On very rare occasions, I will instead get the flashing ERR light instead, but maybe 1 out of 30 times. Used Atmel studio and an AVRISP XPII to program and everything seemed to go fine (no errors). The only difference I saw was that it recognized the power source as 3.1V rather than 3.3V. I have tried programming simple and dynamic programs in Iris, and moving the lpf file to the SD card along with the stock gcal.txt and dc.txt files from the firmware master folder. Usually, after plugging in the power, the LEDs on the board all flash very briefly, then turn off, and the ERR light remains on. Changing the order of events of plugging it in, inserting SD card, etc has not changed the behavior, nor has pressing the reset button (ERR light turns off with push then back on after release).

I am using an alphx innovations 2GB SD memory card, and have tried formatting it in FAT, FAT32, exFAT, and NFTS with no changes. Is there an SD compatibility issue here? We also tried a more current microSD card (in an adapter) with no luck. The SD reader on the board is the replacement one defined in issue #25.

Thanks for your help.

castillohair, 2019-03-08:
Hi @GolgiWhillikers ,

Just to be sure, the appropriate order of events should be either:
- Insert SD Card
- Plug the power connector

or
- Plug power connector
- Insert SD card 
- Press reset button

Is this what you're doing?

Also, did you program the firmware AND fuses of the microcontroller? The full procedure is described here: http://taborlab.github.io/LPA-hardware/initializing/firmware/firmware_programming.html. However, in step e) you should select `EXTXOSC_8MHZ_XX_16KCK_14CK_65MS` instead of `EXTXOSC_8MHZ_XX_1KCK_14CK_65MS` if you have built an LPA v1.1. We currently have a fair amount of the documentation in http://taborlab.github.io/LPA-hardware describing the previous LPAv1.0, but we're in the process of updating it.

The 3.1V measurement worries me a little. If you have access to a multimeter, it would be great if you could use it to double check the voltage in the board.

If all of these are correct and the SD card is still not being recognized, my best guess is that there is a soldering issue somewhere. @KarlGerhardt and I could help you debug this, but it would be better to first rule out the things I described above.

GolgiWhillikers, 2019-03-13:
Well, I seem to have resolved the issue, but I'm not 100% sure what the fix was. I think the most likely scenario is I was trying to load bad Iris programs onto the LPA (the learning curve was steeper than I expected)--I noticed if I plug the SD card in with no lpf program on it, I get the same error light as before. But now, with the SD card formatted in FAT (which is what it was set to by default), with the dc, gcal, and lpf files on the card, it works perfectly fine. Tested 4 SD cards from the same manufacturer and they all worked. I did double check all the flash settings, but we never re-flashed the microcontroller and now it works. So, thanks for the response, seems like everything is good now!

closed 2019-03-13 by castillohair

phuongho43, 2019-03-24:
Did the format of the SD card matter (would it still work with NTFS or exFAT)? Also, are the dc and gcal files in the "Image analysis script" folder? I don't see them in the firmware folder. Can example dc, gcal, program.lpf files be included in the repo?

GolgiWhillikers, 2019-03-24:
I don't 100% understand why it worked after not working for me. The SD card was formatted as FAT (which was the default formatting). There is an example dc and gcal.txt file in the Image analysis folder in the repo which is what I have been using to test my files.

I suspect my problem was that I had not programmed a valid file in Iris, so it didn't work. It would have been useful to have a demo lpf file available for that reason.

castillohair, 2019-03-24:
Hi @GolgiWhillikers  and @phuongho43,

SD cards should be formatted in FAT (aka FAT16) or FAT32, but FAT is preferable. NTFS or exFAT are not supported.

@KarlGerhardt and I are pretty busy for the next couple of weeks, but after that we will release a new version with better file organization and updated documentation. We will make sure to have a set of example files in a more obvious location. For now, you can use dc.txt and gcal.txt from the image analysis folder, and generate a program.lpf file from [Iris](http://iris.taborlab.rice.edu).

phuongho43, 2019-03-25:
It worked! :)
The format was the issue after all. The SD card I got is 64GB and apparently FAT16 has a 4GB limit so I had to make a small enough partition using disk management (Windows 10).