taborlab/LPA-hardware #33: Control LPA via SPI

tlnagy, 2019-03-21:
This is an awesome little board, but my purposes I would love to be able to send signals to the board via serial command. I plan to use the device on a microscope and it would greatly help synchronization if the computer could send a signal that could tell which LED to turn on for how long.

My plan was use an Arduino to send SPI commands to the LPA over the 2x3 SPI pins on the LPA and rewrite the firmware of the LPA to interpret those signals and only turn on the wells that are needed and for the given length of time.

Do you all think this is reasonable? I'm worried that since SPI is already used to control the TLC chips/LEDs that this might not work. I would appreciate your all's thoughts as I'm pretty new to microcontrollers.

castillohair, 2019-04-05:
Hello!

Sorry for the late response. Both Karl and I are deep into thesis defense business these days.

If I'm understanding correctly, you want to use an Arduino to send SPI commands to a free SPI input in the LPA, and rewrite the firmware to use these values to update the LEDs. I think this is feasible. The SD card socket currently uses the SPI module (the LED drivers use the USART module in SPI mode). So you could remove SD card functionality from the firmware and use the now free SPI module for your purposes. The only caveat is that the Slave Select (SS) pin from the SD card module is not available through the 3x2 connector. I think you could get away with soldering a wire from the mircontroller's SS pin to ground, but I would have to look at the datasheet in more detail to be 100% sure.

The other option, if you are willing to do some electronic board redesign, is change the SPI module to manage the LED drivers and use the USART module to communicate directly with the PC without an arduino. This would require rewiring the microcontroller pins and adding a serial connector while removing the SD card connector. This may be a more robust solution depending on how many of these you want to make, how much space you have, and how robust you want it to be. 

Let us know if you have any more questions. I will try to reply much faster this time.

tlnagy, 2019-04-05:
> If I'm understanding correctly, you want to use an Arduino to send SPI commands to a free SPI input in the LPA, and rewrite the firmware to use these values to update the LEDs.

This is correct.

> The other option, if you are willing to do some electronic board redesign, is change the SPI module to manage the LED drivers and use the USART module to communicate directly with the PC without an [...]

This sounds like the approach I would take. What software would you recommend for designing PCBs? I was planning to use Fritzing.

castillohair, 2019-04-08:
I would recommend KiCAD, which is what we used. With it, you should be able to start from our files which should reduce the amount of work you have to do.

Good luck! Please let us know if you have any more questions. And if you're successful, it would be great to have your design available as a fork of our project so that other people with similar needs can use it.