taborlab/LPA-hardware #34: Long Runtime error - Intensity of Lights Change Over Time on Steady State Program

ramininaieni, 2019-06-22:
A long time (~20-30 minutes) after a Steady State Iris program starts to run, the LEDs on our LPA device begin to change in intensity over time. The indicator LEDs on our device throw almost all of the error signals after a long time running the program correctly and flashing green indicating so. The indicator LEDs have thrown error lights indicating "this device has not been able to maintain the specified resolution - ON blinking, FIN off, ERR blinking" & other errors such as "no SD card - ERR on". It would be super helpful if you could let us know if this is an error you have encountered, or if you have any idea what might be causing this error. Thank you!

castillohair, 2019-06-22:
Hello,

How dramatic are the intensity changes? Are they large enough to be visible by eye? We have never seen something like what you're describing. I would first check for proper soldering of all pads. If you have a small microscope or amplifying lens, you could use it to spot any places where there is not enough solder or two adjacent pins are incorrectly connected. If all soldering is correct, I would try replacing the microcontroller or the voltage regulator chip.

ramininaieni, 2019-06-24:
Hi there,
Thank you so much for your prompt response! Yes, the changes are very visible by eye. It must be a problem with the soldering or the parts on the board if you guys haven't seen this problem before. I will let you know if this fixes the problem! Thank you!
-Roya

ramininaieni, 2019-08-08:
Hi there, 
Soldering a new board with a microscope and with new parts appears to have fixed the problem! Thank you so much for your advice & help!
Sincerely,
Roya

closed 2019-08-08 by ramininaieni