taborlab/LPA-hardware #35: Other potential 24 well TC treated plates to use with LPA?

ramininaieni, 2019-07-22:
Hi there! 

I am wondering if there are alternative vendors to get plates like https://arcticwhiteusa.com/product/awls-303008 (TC version: AWLS-303012) for use with the LPA. I am asking because Arctic White does not appear to be a conventional vendor. Are they the only vendors that sell plates that can be used with the LPA? Thank you!!

Best,
Roya

castillohair, 2019-07-23:
I have recently searched for cheaper equivalent plates, but to my frustration the only thing I was able to find was equivalent plates sold by other vendors. This seems to be one example https://www.thomassci.com/Laboratory-Supplies/Plates/_/24-Well-Krystal-Clear-Bottom-Polystyrene-Assay-Plates. The part number is exactly the same, as is the price if I'm not mistaken. I recommend you to request the dimensions and compare them to the ones from Arctic White before ordering. Other vendors can be found by googling "24-well Krystal™ Clear Bottom Polystyrene Assay Plates".

ramininaieni, 2019-07-24:
[page 28 of the Porvair catalog.pdf](https://github.com/taborlab/LPA-hardware/files/3428777/page.28.of.the.Porvair.catalog.pdf)

I reached out to Thomas Sci and they said that they were the same plates as the ones sold by Arctic White. To make sure, I checked the dimensions (attached above) against those of the plates used for the LPA to make sure and the plates sold by Thomas Sci do appear to be the same ones sold by Arctic White and used for the LPA. Thanks so much for all your help! I really appreciate it!

closed 2019-07-24 by ramininaieni