taborlab/LPA-hardware #36: How was such a low initial coefficient of variation of 7.78% achieved in LED Calibration?

ramininaieni, 2019-07-25:
Hi there!
So, for some reason my initial coefficient of variation (sd/mean) when calibrating my LEDs to each other on my board is 40%. After 3 rounds, it is reduced to 4%. This is regarding the gel imager method for single wavelength lights. I am using the same blue LEDs as those used by the Tabor lab in https://www.nature.com/articles/srep35363 (*RL5-B2545; ID: 470-SB*)

I notice that in the example provided: http://taborlab.github.io/LPA-hardware/initializing/calibration_led.html the initial coefficient of variation is 7.78% and is reduced to .55%. I also notice that in the matlab analysis, red (channel 1) and green (channel 2) are used. 

In the example linked above that you provided, are you using red/green LEDs? Have you noticed a lot of/more variation in the blue LEDs that you use than the red ones? 

I'm just trying to figure out why there is so much initial variation in my LEDs compared to the example and I am wondering if this is because the example is for a different wavelength LED and maybe the blue LEDs used in your research paper just have a lot/more initial variation. 

Thank you soooo much for all of your help. We are really excited to start using the LPA :D 

Best,
Roya

castillohair, 2019-07-29:
Hi Roya,

Sorry for the delay. I was out at a conference.

Maybe @KarlGerhardt can give you more specific answers, but I would not try to read too much on the initial variance of LED brightness. LED variance is dependent on several factors:
- intra-batch variability: if you buy 10 LEDs, they're obviously not going to have the exact emission features. And probably some manufacturers are more consistent than others.
- inter-batch variability: if you buy several batches of LEDs over a long period of time, they're probably gonna have even more different emission characteristics.
- usage: we have found noticeable degradation in light intensity in LEDs that have been heavily used. You will see even greater variability if you're combining old with new LEDs, or old LEDs that have been used unevenly.

Therefore, my recommendation is to not stress too much about initial LED variability, and just try to bring it down as much as you can. A few percentage points seems acceptable in my opinion.

ramininaieni, 2019-08-08:
Hi there,
Thank you so much for your advice. We've decided that a few percentage points (CV of 4%) is okay for our needs. 
Sincerely,
Roya Amini-Naieni

KarlGerhardt, 2019-08-09:
Hi Roya,

40% seems pretty high. It's been a while since I calibrated those LEDs, but I vaguely remember them having lots of "structure" (spatial non-uniformity) in the light output. For LEDs with lots of structure, using diffuser paper is crucial. 

You also want to make sure that you are not using low Iris or gcal values (e.g. < 100) when performing the calibration. I wouldn't predict this to cause high initial CV, but it can limit how low you can get the CV by calibrating.

Also, check out the recent MIE chapter [MIE chapter](https://www.sciencedirect.com/science/article/pii/S0076687919301338) which has detailed protocols.