taborlab/LPA-hardware #37: Programming issues

morphogenetic2, 2022-02-17:
Hi there, firstly many thanks for making this fully open source. I just built a LPA following your instructions and I must say that my experience with electronics is very limited. I soldered the components manually (I don't have access to an oven) and double checked every component and location and seems fine, but when I try to program it, I'm getting a low voltage warning and the programmer is not recongnizing the ATMEGA. I just wanted to know if, besides the LEDs, electrolytic capacitors and obviously the chip/LED drivers/voltage regulator, schottky diode and Xtal, if there is any other component with requires an specific polarity, like the inductor or the ceramic capacitors. 

When I plug in the power supply (5V, positive barrel), is any of the SMD LEDs supposed to light up? Is there any specific connection order to program the microcontroller? Should LEDs be already plugged in in the sockets?

Sorry for many questions, but I don't have enough expertise to troubleshoot this myself, many thanks for your time.

castillohair, 2022-02-17:
Hi. Just to confirm a few things: Are you using the 1.0 design (the one with the molex connector as in figure 1 of https://www.nature.com/articles/srep35363) or the 1.1 design (the one with the barrel jack connector as in figure 2 of https://www.sciencedirect.com/science/article/abs/pii/S0076687919301338)? When you get a "low voltage" warning, does it show a voltage number? What power supply are you using? If you're using the 1.1 design, can you make sure the breakaway headers are soldered in the correct orientation (i.e. pointing in the same direction as the LED sockets, see figure 2 of https://www.sciencedirect.com/science/article/abs/pii/S0076687919301338)?.

If you have access to a multimeter, the next step would be to measure a few nodes to make sure all the voltages are correct. Also let me know if you have access to an oscilloscope. If you have access to a simple microscope, it might also be a good idea to check the soldering junctions with it. We have found false contacts or crossed junctions in other people's devices previously that weren't obvious by eye.

morphogenetic2, 2022-02-18:
Hi @castillohair thanks for the quick reply. I'm using the 1.1 design. The warning I get is that the voltage is low (0.2 or 0.3V) and that the Atmel studio is unable to recognize the device. I'm using a 5V, 2.5A, 12.5W adapter (RS online UK, P/N 175-3302), with a standard 2 mm barrel adapter (internal positive and external negative polarity). The headers are soldered in the correct orientation.

I do have access to a microscope, a multimeter and an oscilloscope. I soldered all the parts using the microscope but I'm concerned about "frying" some components due to using excessive heating, is that possible?

What nodes should I check?

Thanks very much again.

castillohair, 2022-02-18:
Good. We should start by making sure the ICs are receiving appropriate voltages. Please measure DC voltage with the multimeter between pins 1 and 4 of the AP1509 power regulator, and also between pins 3 and 4. This should be done with the power supply connected. The first one should be 5V, and the second should be 3.3V.  Be very careful to not short adjacent pins with the multimeter probes. If these measurements are successful, next I would measure voltage between pins 4 and 5 of the ATMega, which should also be 3.3V.

For further reference, I think the most up-to-date circuit diagrams of v1.1 are in my thesis https://scholarship.rice.edu/handle/1911/106008, starting from page 23.

morphogenetic2, 2022-02-23:
Hi @castillohair, I measured the voltages with a multimeter and were 5V for the jack barrel (power supply OK I guess) but zero on the voltage regulator and the IC. So I went again to check the soldering and looked clean, so finally decided to swap the voltage regulator with another one I had around, and this time worked like a charm, the voltages were correct and managed to program the ATMega with your provided firmware. Thanks so much for your support :)

castillohair, 2022-02-24:
Glad it worked out!

closed 2022-02-24 by castillohair