taborlab/LPA-hardware #38: Microcontoller availability

lanmasterlan, 2022-03-26:
Hello,

The availability of the ATMEGA328-AU is currently very bad. Can an ATMEGA328P-AU be used instead? If so, are there any firmware-related changes required?

Thank you.

castillohair, 2022-03-27:
Hi,

Thanks for your question. Unfortunately I can't 100% guarantee that they're interchangeable within the design of the LPA because I haven't tested it myself. However, from a quick search, it seems that the 328P would work fine in place of the 328. You might need to recompile the firmware though, while selecting the appropriate chip. You can check the paper's SI for more info on how to do this. If you have issues while doing this, please let us know and we can go from there.

Hope this helps.

lanmasterlan, 2022-03-29:
Thank you very much for your help.
I managed to recompile the firmware. I used the most recent version of Microchip Studio and generated libcore.a using the old Arduino IDE version 1.6.5. I adjusted the licore.a path and then changed ATmega328 to ATmega328P under Project --> Properties --> Device. I was then able to run Rebuild Solution and got the firmware.elf file, however with the warnings attached below.

Will it be okay?

Severity | Code | Description | Project | File | Line
-- | -- | -- | -- | -- | --
Warning |   | unused variable 'n' [-Wunused-variable] | firmware | C:\Program Files (x86)\Arduino\libraries\SD\src\utility\Sd2Card.cpp | 379
Warning |   | logical not is only applied to the left hand side of comparison [-Wlogical-not-parentheses] | firmware | C:\Users\user\Downloads\LPA-hardware-1.11\Firmware\Source\firmware.cpp | 42
Message |   | in expansion of macro 'System_IsState' | firmware | C:\Users\user\Downloads\LPA-hardware-1.11\Firmware\Source\firmware.cpp | 400

castillohair, 2022-03-29:
Great! I don't think those messages would be an issue.