taborlab/LPA-hardware #39: Abnormal behavior after programming

RZHOU-UT, 2022-04-22:
Hi, 

I am unable to get the LPA to function properly after programming and I am unsure what the source of the issue is. I am an undergraduate with little electronics experience so troubleshooting has been difficult. I have consulted with some professors and graduate students and have still been unable to resolve the issue. I will try to describe the issue here.

The 48 LEDs do not light up at all. I am unable to determine the error based on the 3 indicator LEDs because all 3 indicator LEDs are turned on, which is not an error listed in the supplemental material. Additionally, the indicator LEDs occasionally will behave erratically (all three will blink, some combination will light up unexpectedly, etc.). The one condition in which the indicator LEDs behave properly is when the SD card is removed, though the board takes a while to respond. 

I have checked the hardware and all connections appear to be okay. However, when I program the device, there are two areas of concern. 

1. The target voltage I am reading is 4.7 V instead of the 3.3 V that should be expected based on the programming instructions. What does target voltage refer to? We are using an AVRISP 2.0. The board is powered by a 5.2 V power source. The Vout of the voltage regulator is around 3.15 V, and this goes to the Vcc of the microcontroller. I have just seen the previous issue opened on github about target voltage and will also check the orientation of my programmer (maybe the red line does not face out with this programmer). 
2. When I program the chip, it says that programming flash is okay. Afterwards, it says verifying flash...failed. What might be the source of this error? 

Thank you for your time and help. I believe we are using an LPA 1.0. Please let me know if there is any other information or images I can provide.

castillohair, 2022-04-22:
Hello,

If you have measured 3.15V coming out of the regulator but the AVRISP 2.0 is detecting 4.7V, it sounds like the problem might be the AVRISP or the connections to it. Note that the orientation of the programming pins are different in the LPA v 1.0 compared to v1.1. If you could send us a picture of how you soldered these pins and how you are connecting the AVRISP to it, that would be very helpful.

Thanks.

ri-zhou, 2022-04-22:
Here are some images.

[image]
[image]

What is target voltage? How does target voltage affect programming?

There are a few more things I want to note.
1. The programmer we are using: https://wiki.seeedstudio.com/Atmel_AVRISP_STK500_USB_ISP_Programmer/. This programmer does not list the ATMEGA328 in compatible chips, though it lists the ATMEGA328P which seems to be essentially the same thing. 
2. Our AC to DC adapter outputs 5.2 V instead of 5, even though it was advertised as a 5V adapter. I do not think this is an issue, but if it is, let us know. 

By the way, the second issue I mentioned in the original post (verifying flash...failed) was resolved when I changed the CKSEL to 16K CK/14CK from 1K CK/14CK, even though I am using an LPA v1.0. 

Thank you for the quick response! Unfortunately, I think there is also a hardware issue as we may have damaged the voltage regulator. The measurement of 3.15 V we made was not typical. It seems that now the Vout of the regulator is typically around 2.6 V, which is an issue because the LED driver requires Vcc of 3.0 to 5.5 V. Still let me know what you think about the programmer, target voltage, and the indicator LED behavior, but it will be a few days before we can acquire a new voltage regulator and try to see if that resolves any issues.

castillohair, 2022-04-28:
Hi,

Sorry for the delayed response. You are right that 2.6V will not be enough to power the microcontroller. Did you get to replace the regulator? 5.2V at the input is fine. The only hard voltage requirement is 3.3V for the microcontroller and LED drivers, but the regulator normally takes care of that. 

Also, are you programming the firmware with the SD card inside? The SD card and the programming pins come from the same place in the microcontroller, so there might be some interference there. I recommend trying to program the firmware with no SD card inserted into the board.

I would think the programmer would be compatible with all variations of the ATMega328 but I can't be 100% sure since I haven't used that particular one. I'm also not 100% sure about the SUT_CKSEL flag but after a quick inspection of the datasheet it seems to me like it would be fine.

Hopefully some of this helps. Please keep us updated!

ri-zhou, 2022-05-02:
The voltage regulator was replaced and it functions as expected now. I have not connected the programmer again and tried to read the target voltage (though it may have changed to 3.3V as expected). Thank you for your time helping with this. The voltage regulator was replaced and it functions as expected now. We appreciate your responsiveness and we hope to use the LPA soon.

closed 2022-05-02 by ri-zhou